sufst/vcu #26: Inter-thread messaging timeout behaviour

t-bre, 2022-02-05:


t-bre, 2022-02-09:
@RoryWilson92 something else to think about for the queue system. Not sure what good options we have in this scenario to be honest.

closed 2022-05-24 by t-bre

RoryWilson92, 2022-05-24:
> Closed #26 <https://github.com/sufst/vcu/issues/26> as completed via #113

Wait is this done now ? I haven't been working on it due to exam season but
I intended on finishing it off ? or is it too late?

t-bre, 2022-05-24:
@RoryWilson92 Wasn't sure if you were coming back at any point. I guess to give you an update - only one thread ever sends anything to the CAN thread at the moment (torque requests), so there isn't a huge benefit in a priority queue system this year. It should still be useful in future though, and it's an interesting problem.

That said, I ended up coming up with a much simpler way to do priority queues when I was writing some other code. The gist of it is:

```C
QUEUE high_priority_queue;
QUEUE low_priority_queue;
SEMAPHORE priority_queue_semaphore;

function priority_queue_send(message, priority)
{
    if (priority == HIGH)
    {
        queue_send(high_priority_queue, message);
    }  
    else
    {
        queue_send(low_priority_queue, message);
    }

    semaphore_post(priority_queue_semaphore);
}

function priority_queue_receive(message_ptr)
{
    semaphore_get(priority_queue_semaphore);
 
    if (!queue_empty(high_priority_queue))
    {
        queue_receive(high_priority_queue, message_ptr);
    }
    else if (!queue_empty(low_priority_queue))
    {
        queue_receive(low_priority_queue, message_ptr);
    }
}
```

Not actual code, but you get the idea. Should be fairly scaleable given the simplicity unless there were a high number of priority levels. If you still want to work on this you can, but regular weekly meetings have ended and we're entering the build phase which means it's realistically too late for this year.

PR #113 (2022-05-24): Simplify message passing system
### Changes
- Removed old half-complete message passing code.
- Implemented same functionality using standard ThreadX queues.

### Fixed Issues
Closes #26 
Closes #112 

### Other Related Issues
#31