sufst/vcu #54: Fault state

t-bre, 2022-03-15:
Possible causes:
- Safety critical signals out of range for more than 100ms.
- 10% deviation in expected throttle pedal input difference.
- Inverter?

In fault state:
- Digital output signal needs to be driven low #53.
- Flash red LED.
- Do not leave fault state until power cycled.

Implementation:
- Watchdog timer?
- Thread entered on fault with maximum priority.

lebedys, 2022-03-16:
FSUK Rules (2022, v1.0)

**EV 4.11.6** The vehicle is ready-to-drive as soon as the motor(s) will respond to the input of the APPS.

**EV 4.11.7** After the TS has been activated, additional actions must be required by the driver to set the
vehicle to ready-to-drive mode (e.g. pressing a dedicated start button). The transition to
ready-to-drive mode must only be possible during the actuation of the mechanical brakes
and a simultaneous dedicated additional action.

**EV 4.11.8** The ready-to-drive mode must be left immediately when the shutdown circuit is opened.

lebedys, 2022-03-19:
This fault state must be entered when a digital input (3V3) is low.

t-bre, 2022-03-19:
> This fault state must be entered when a digital input (3V3) is low.

I assume this signal comes from some other circuit / subsystem? So we would just need to monitor it continuously (say checking once every few RTOS ticks).

t-bre, 2022-06-14:
This was finished by various PRs.