sufst/vcu #82: Inverter speed requests

t-bre, 2022-05-06:
### Description
Option to substitute torque requests for equivalent speed requests.

### Specification
- [ ] Config file `#define` flag to activate this mode (@t-bre).
- [ ] Config file check to disallow this mode in competition/release build (@t-bre).
- [ ] Speed request CAN interface function (@hashyaha).
- [ ] Integrate flag check and interface function into CAN thread (@t-bre).

Function prototype for speed request CAN interface:
```C
pm100_status_t pm100_speed_command_tx(UINT speed);
```

### Additional context
This is needed for the motor test rig which has no loading. Making torque requests under no load is generally a bad idea!

t-bre, 2022-05-06:
@maxobrien The control thread would still generate a desired torque in this scenario, mapped from the throttle pedal through the driver profile. Do you think that should just directly map to a speed request (as in `UINT speed_request = torque_request`).

hashyaha, 2022-05-07:
I can use the Speed Mode Enable bit in command message to override the inverter to speed mode. 
Anyway, it might be useful to know:
> The output of the speed regulator is a torque command.
Because the speed regulator can drive a motor to the maximum torque it must be
used with caution. If for example the resolver is not calibrated a motor may not
make any torque even with a large amount of current flowing. Thus the motor
could quickly to maximum current without any movement of the motor.

taken from [Using Speed Mode](https://app.box.com/s/vf9259qlaadhzxqiqrt5cco8xpsn84hk/file/49688273726)

t-bre, 2022-05-09:
Is it possible to set a current limit on the inverter itself?

hashyaha, 2022-05-09:
Yes, the current limit can be set for Q and D-axis before motor is enabled. The current limit takes precedence over torque command and torque limit: 
> However, if the current limit of the drive is reached before the torque command has been achieved the controller will limit on the current first. If this happens the operator will feel an additional amount of unused pedal range at the top end. 
(Pg 48 of [Cascadia Motion CAN Protocol](https://app.box.com/s/vf9259qlaadhzxqiqrt5cco8xpsn84hk/file/27334613044))

closed 2022-05-12 by t-bre