sufst/vcu #88: Conflicting GPIO pin PB14

t-bre, 2022-05-12:
### Description
`CAN_Rx()` in `can_types.c` writes to GPIOB pin 14. This pin is already assigned to the built-in red status LED.

### Expected behaviour
Different pins should be used to avoid conflict.

### Additional context
Start of `CAN_Rx()` function:

```C
void CAN_Rx()
{
    FDCAN_RxHeaderTypeDef RxHeader;
    uint8_t	RxData[8];

    queue_msg_t Rx_msg;
    HAL_GPIO_TogglePin(GPIOB, GPIO_PIN_14);

    // ...
```
Conflicting lines in `main.h`:
```C
#define RED_LED_Pin GPIO_PIN_14
#define RED_LED_GPIO_Port GPIOB
```

t-bre, 2022-05-12:
@hashyaha do you known the purpose of the `HAL_GPIO_TogglePin()` call here? Is it related to CAN functionality or is it intentionally toggling the LED for each CAN transmission?

hashyaha, 2022-05-12:
Yea, you're right, I was using it as debug LED. It should be removed.

t-bre, 2022-05-12:
Great thank you, I'll do that 👍

t-bre, 2022-05-12:
Is it still useful to have a debug LED? I could move it to the yellow LED which is unused?

hashyaha, 2022-05-12:
That would be great :)

PR #93 (2022-05-12): Fix GPIO LED pin conflict in CAN driver
- CAN driver now uses yellow LED for debug instead of red LED which was already reserved for fault indication.
- LED toggles every time `CAN_Rx()` is called.
Closes #88
closed 2022-05-12 by t-bre