sufst/vcu #94: Inverter CAN broadcast configuration

t-bre, 2022-05-13:
### Description
We only want some of the information which the inverter broadcasts. Broadcasting of information which we don't need should be disabled to reduce traffic on the CAN bus.

### Specification
- [ ] Work out how to set which information is broadcast (see datasheet).
- [ ] Create function to send parameter write messages to the inverter to enable/disable a particular broadcast.
- [ ] Create parameters in configuration header which can be set to enable/disable each broadcast message. Possibly simplest to enable/disable in groups matching the datasheet.

### Additional context
[Cascadia Motion CAN Protocol](https://app.box.com/s/vf9259qlaadhzxqiqrt5cco8xpsn84hk/file/27334613044)

t-bre, 2022-05-13:
@hashyaha I've constructed the bytes which need to be sent to configure broadcasting of CAN messages. Should this be set via CAN with `pm100_eeprom_write_blocking()`? There are four bytes of data which need to be sent (page 19 of datasheet, both upper words), but at the moment this function only transmits 2 bytes of data.

hashyaha, 2022-05-13:
I think there might be discrepancy in Page 19 because in pg 44 there are two EEPROM address corresponding with CAN broadcast Hi word (237) and Lo word (148). This matches the description of parameter message in pg 38 which has 2 data bytes only.
[image]

 In this case I think it may be sufficient to write to Lo word, as it is not recommended to change the default Hi word (0xFFFF).

PR #95 (2022-05-15): Inverter broadcast configuration
- Configuration header settings to enable desired types of inverter broadcast messages.
- Construction of lo word for broadcast configuration parameter command message.
- Transmission of broadcast configuration command on initialisation.

Closes #94
closed 2022-05-15 by t-bre