sufst/vcu #183: chore(vscode): work out how to generate compile commands database on Windows

t-bre, 2022-11-15:
### Description
Work out how to generate compile commands database on Windows.

Possibly using [ccdgen](https://github.com/t-bre/ccdgen) which is what is used on macOS.

jammyb1, 2022-12-11:
> @jammyb1 @Scaniox you guys have used `ccdgen` for this, would you say it works or are there some problems still to sort?

Yeah I've been using it and it works fine 👍

t-bre, 2022-12-11:
> It works well. Only hiccup is that it doesn't work If the build directory is already there (clean sorts it out)

This should be fixed on `develop`, I changed the makefile to use `mkdir -p` to create the build directory. This should not cause an error if the directory already exists.

t-bre, 2022-12-11:
With that I think this is closed then!

closed 2022-12-11 by t-bre