sufst/vcu #250: feat: Torque mapping based on battery charge

Avdieienko, 2025-01-26:
### Description
When the battery is getting close to nearly empty, pulling 80kW would result in a voltage drop so big that it would shut down the inverter (it has a 300V min voltage). Need to implement some throttle pedal mapping that decreases the maximum allowed torque request as the SoC(State of Charge) signal decreases. Until tests (to get proper formulas and limits) we can just linearly limit torque after 60%.

### Requirements

- [ ] Read `BMS_Pack_SOC` reading
- [ ] Implement additional mapping in both `config.c` and `torque_map.c`

### Additional Notes
`BMS_Pack_SOC` signal is in `MSGID_0x201` message in `CAN-S`

r-kirkbride, 2025-03-25:
suggestion: add function to subscribe passing a pointer in

Avdieienko, 2025-03-29:
@r-kirkbride we did some battery simulations and we might limit torque based on the temperature of the battery instead of SoC (or with both SoC and temperature) to reduce current sent and stop the battery from heating up(BMS fault and shutdown closed after it is overheated).

[image]

r-kirkbride, 2025-04-29:
Max temp before fault? 60

Max temp before limting torque?
Temp for max limting?
Max torque limit?

Avdieienko, 2025-04-29:
@r-kirkbride
Max torque - 220
Min torque - 50
Start from 50 degrees
Linear

r-kirkbride, 2025-04-29:
What temperature should it reach the min torque? The temperature fault threshold?